naikymen/klipper-for-cnc #9: Project alive? And RS485 spindles support.

g992, 2023-06-09:
Hello! is the project still supported? and is it possible to connect rs 485 controlled spindle?

naikymen, 2023-06-28:
Hi! Yes it is alive :)

I'm not sure that spindles are properly supported by (upstream) Klipper yet. Have you already tried to ask this question at the Klipper forum?

This is why I had not even considered spindles yet, but I just found that this has been discussed in the forum at least once, and a rudimentary extras module for spindle control has been written: https://klipper.discourse.group/t/klipper-cnc-spindle-rpm/8318

I imagine that getting precise spindle speed control would require a massive amount of work on the firmware side of Klipper, but I'm not savvy enough to give it a try.

Contributions would be very welcome!

naikymen, 2023-06-28:
This might be of use to you as well: https://github.com/Klipper3d/klipper/pull/4128

Fast PWM updates are likely useful for spindle control.

g992, 2023-07-24:
and i found this https://www.klipper3d.org/RPi_microcontroller.html
what about useing rpi2040 for pwm generation?

naikymen, 2023-12-19:
Hello again! Sorry I missed your message entirely. Klipper has implemented proper PWM for tools (lasers, spindles, etc) upstream, and I have already merged it to this branch.

Does that feature fullfil your requirement?

naikymen, 2024-05-06:
Closing due to inacivity.

closed 2024-05-06 by naikymen