naikymen/klipper-for-cnc #12: [feat] Adapt additional kinematics - Post your wish here!

naikymen, 2023-12-19:
I only have a cartesian machine, with linear ABC axes.

If you own, say, a delta or polar machine, let me know here and I'll try adapting it when I find the time.

Ravs1999, 2023-12-20:
Hi naikymen. Really appreciate your work on expanding Klipper to more than 3 axis! I have a mechanical/electrical background so not too proficient in programming but i know the basics in C. I tried reading through and understand your code but was wondering if your fork of Klipper would be suitable for my needs. 

My team and I are developing toolhead where we have a primary filamen extruder but then we also have another mechanism which rotates around the central nozzle. The purpose of this rotating mechanism is to maintain a "tangency" to the XY toolpath. Pretty much like a tangential cutting knife but we cant repurpose the extruder axis as we need the extruder + this rotating mechanism. I was thinking that this would be a "4th axis" in addition to XYZ. The extruder would follow normal 3 axis kinematics and the rotating part just rotates so it maintains a tangent to the toolpath Obviously, we would require the axis to be in sync with XYZ motion. Would something like this be possible with your fork? I've already done it with DUETs RepRap firmware but it is easy since they allow for additional, linear syncronized axes.

Ravs1999, 2023-12-20:
> Hi naikymen. Really appreciate your work on expanding Klipper to more than 3 axis! I have a mechanical/electrical background so not too proficient in programming but i know the basics in C. I tried [...]

So a typical G1 command would be G1 X10 Y10 Z10 Aa where A is rotatational axis and a is a corresponding rotation amount

alkis01, 2024-05-06:
Hybrid CoreXY please for use with idex printers. Klipper allows only one z-probe. Having the ability to use multiple probes (probe_G38_multi)  will be great as you can have one on each of the 2 toolheads and automatically calculate their offset.

noobydp, 2024-05-06:
@alkis01 you shouldn't need a probe on every tool. 

An alternate approach is to have a probe on the carriage or T0, use that for gantry levelling/mesh, and then measure the z offset for each other tool. 

There are some macros/plugins to do this, but I haven't used them (yet). E.g. https://github.com/protoloft/klipper_z_calibration

alkis01, 2024-05-06:
@noobydp thank you for the link. My case is a bit different, I have an idex with 2 TAP toolheads. I do the leveling/mesh with T0. The idea is after homing and leveling to use T1 and measure distance from the bed so I can calculate the offset between the 2 nozzles. I haven't found how to do that in Vanilla Klipper, so I though using the G38 and giving as a switch pin my T1 TAP might work.

noobydp, 2024-05-06:
@alkis01 Woah. That sounds awesome. 

Photos? 😬

naikymen, 2024-05-06:
> Hybrid CoreXY please for use with idex printers. Klipper allows only one z-probe. Having the ability to use multiple probes (probe_G38_multi) will be great as you can have one on each of the 2 toolheads and automatically calculate their offset.

I think I'll need some education before taking that up, but its on my to-do list now :)

Send pics!

vtpgit, 2025-07-05:
Hi there, awesome work! Question, not sure if this is the place. Does your fork support stall-guard monitoring of the diagnostic pins in real-time for tmc5160 in order to detect obstacles to protect the frame? Cheers!