naikymen/klipper-for-cnc #13: Looking for printer.toolhead.homed_axis to include 'A,B,C'

adjustinthings, 2023-12-22:
I'm using moonnode in nodered to pull data in from moonraker for my robot automation project. I just switched to this klipper for cnc and it exactly what i need. 

It seems printer.toolhead.homed_axes doesn't display the A,B,C axis as homed. Is there another location that information is stored? 

OR is my moonraker just not able to get that info? (moonraker v0.8.0-209-g42357891)

I'm looking through cartesian_abc.py for clues currently.

adjustinthings, 2023-12-23:
[image]
oh i see

naikymen, 2023-12-24:
Hey! Thanks for posting.

I completely forgot about the status stuff.

I just pushed a commit that should at least get the info out there: https://github.com/naikymen/klipper-for-cnc/commit/694f919b0cb886797ed22f52464249d89e818c44

Added also a `GET_STATUS_MSG` command which might help debug stuff:

[image]

naikymen, 2023-12-24:
Please let me know if this is what you needed and how it goes :)

naikymen, 2023-12-24:
Also added the extruder's status, in case it's home-able: https://github.com/naikymen/klipper-for-cnc/commit/87db854eadb226f64bfeaee4132a78dbe42e099a

Its messy but it seems to work.

adjustinthings, 2023-12-25:
Hell yes. You rock, thank you. Works for me. I can see all the info in the printer.toolhead 👍. I'm using it to make a gantry automation system.

naikymen, 2024-05-03:
Fantastic! Glad that it works now.

Could you share an image of your machine? Just curious :robot:

closed 2024-05-03 by naikymen

adjustinthings, 2024-10-13:
> @adjustinthings <https://github.com/adjustinthings> Could you share an

I will get you this photo. I was trying to clean stuff up a bit. Its a
gantry system that attaches to the front of racking to pull things off the
shelf and move trays around.