naikymen/klipper-for-cnc #18: Homing of A Axis does not work

Thomas-Toucan3d, 2024-04-13:
[klippy.log](https://github.com/naikymen/klipper-for-cnc/files/14967620/klippy.log)

Hi Naikymen,
I am trying to configure a 4-axis printer using XYZ + A. I think I have configured everything correctly, but unfortunately the A axis does not home. I have also looked at your examples of emplementation and tried several things, but I have not been able to solve the problem. 
[image]

Do you have any ideas what the problem might be? I have attached the log file where I homed the printer once. 

Thank you for your time and help! Let me know if you need more information and I will do my best to provide it!
[image]

Best,
Thomas

noobydp, 2024-04-14:
It looks like there's something wrong with homing A when you use [safe_z_home] in your config.

I did a test with my machine and a rotary axis A, when safe_z_home is defined it will home XYZ but doesn't do anything when I run "G28 A". The A axis doesnt move and there are no errors, is this what you experienced?

Can you remove the [safe_z_home] section, possibly replace it with a custom G28 homing macro?

naikymen, 2024-05-03:
Hello! Thanks for reporting and finding a workaround. I'll let you know when I fix the issue.

naikymen, 2024-05-18:
Hi, I've implemented some basic support for multiaxis in safe_z_home. Please test it with your fingers on the E-stop button as usual.

I didn't have time today for a hardware test.

naikymen, 2024-09-18:
Hi all, I'll be closing this soon if there are no objections. Hoping everything is working now.

closed 2024-10-16 by naikymen