naikymen/klipper-for-cnc #21: Implement "emergency brakes" in order to stop but also retain position

naikymen, 2024-05-19:
`EMERGENCY_STOP` is too violent for my taste.

An `EMERGENCY_BRAKE` would stop the machine in the way jogging would (according to my idea of that feature https://github.com/naikymen/klipper-for-cnc/issues/20).

The emergency brake would signal the MCUs to ignore all incoming commands (entering LOCKED mode), and apply a brake move at maximum acceleration. Also clear all move queues immediately everywhere. There must be a new command that UNLOCKS the machine after an emergency brake.

naikymen, 2024-05-21:
This is only meant to retain the robot's position in an emergency, and using that information to avoid collisions while re-homing.

I'd consider it an emergency function, but its logic might be good for restarting from a certain point too, but it will need extra thought.

I'm sure a GCODE program could be restarted from this "soft-stop" state, by using the print time and position/step at the soft-stop event, and some extra context from the host part.

I don't know how to implement any of this in the MCU code, at all, but someone with some experience would not find it hard to do. Hence the "help wanted" tag. :smile:

john-clark, 2026-02-24:
Maybe more like a pause?