naikymen/klipper-for-cnc #24: simultaneous xy homing

nikewua, 2024-06-17:
Is it possible to make the X & Y axis home simultaneously?
As an example (for core xy kinematics) the toolhead would move diagonally towards the 0,0 position and home whichever endstop is reached first then continue to home the second axis.

naikymen, 2024-06-18:
Hi thanks for bringing this up. I think it has been discussed in the Klipper forums already, but it's been a while since I searched. Have gone through the forums/discord lately looking for updates on this?

> whichever endstop is reached first

I suspect that the Klipper firmware does not support that in particular. AFAIK, all "homing" moves are tied to one single endstop. I do not know about combining endstops with that kind of and/or logic with the current firmware.

Perhaps someone with more wisdom on the firmware side can chime in.

nikewua, 2024-06-19:
> Hi thanks for bringing this up. I think it has been discussed in the Klipper forums already, but it's been a while since I searched. Have gone through the forums/discord lately looking for updates on this?
> 
> > whichever endstop is reached first
> 
> I suspect that the Klipper firmware does not support that in particular. AFAIK, all "homing" moves are tied to one single endstop. I have not heard about combining endstops with that kind of and/or logic with the current firmware.
> 
> It would also be weird to implement from the host side only. I don't know if it is possible to execute two "drip moves" at the same time.
> 
> Perhaps someone with more wisdom on the firmware side can chime in.

Thank you. Indeed, the klipper firmware does not support it. My machine has two components that must be homed simultaneously, so I came to seek help