naikymen/klipper-for-cnc #26: Support for UVW axis

JordyDH, 2024-09-06:
We are building a PNP powered by klipper but we need XYZABCUVW control.

In the code i see some mention of the UVW axis but i think it is not implemented. 

Is it on the roadmap to implement this? Or can you share what needs to be done for this so maybe me and my team can have a shot at this?

naikymen, 2024-09-06:
Interesting, what electronics are you using? Is it an open hardware machine?

I have tried to make my modifications rather modular, so its possible that adding 3 more axes is not much work (for me at least). This means that the code is still hardcoded to either XYZ only (most upstream code), or XYZABC only (this fork), in a few places. I would not expect any major refactoring.

Again, most of the code should respond to adding an extra "uvw" axis set, and I have added _many_ comments explaining what I did.

If you want to study the source, I'd recommend that you start reading the toolhead's init (https://github.com/naikymen/klipper-for-cnc/blob/pipetting/klippy/toolhead.py#L432). Also have a look at load_axes, setup_kinematics, in the toolhead.py module. Everything else will need reviewing too.

Also consider that having 9 axis sets, with 3 kinematics in total, may end up slowing klippy down, maybe enough to start causing timing problems (especially on a raspberry pi).

I'm rather short on free time, so you're welcome to give it a go. :smile:

There is no roadmap, but this feature would be in one of course.

JordyDH, 2024-09-07:
@naikymen We have recently bought a Neoden4 and have removed all the control electronics from it.
Currently, we are upgrading it to a Klipper setup with a BTT board. We have discovered that the original neoden pcb board in the head uses an MCU that is supported by klipper, and they also use can-bus for the communication.

We have fully reversed engineered the head pcb, only the mcu is locked to read from it. Just to have a backup, we are going to solder a new MCU on it to flash with klipper.

Are you going to actively develop on this fork to be up-to-date with the progress of klipper?
Or is this just a fork to show what is capable?

I have looked now into the code and (I think) I know how to implement the UVW axis, fingers crossed.

naikymen, 2024-09-07:
Looks like fun!

Yes I'll keep this fork up-to-date with upstream Klipper. I rely on it for a lab automation project.

Please consider opening a pull request, I might find time to lend a hand or review ideas, and would like to see what you come up with.

Good luck!

JordyDH, 2024-09-09:
https://github.com/naikymen/klipper-for-cnc/pull/28

I have added the support for UVW axis.
It is still hard coded, but I'm experimenting with a generic piece of code that is able to extend up to 26 axis.
This would probably be overkill, but it would be nice to just have the flexibility.

naikymen, 2024-10-01:
Hi again how is it going? Any updates on this front?

JordyDH, 2024-12-14:
Hi

We recently got our package from bigtreetech so we can continue soon the progress. We are right now upgrading the machine with a better board and tmc5160 drives to really push the motion system (60v 8-10A).