naikymen/klipper-for-cnc #31: [Fork] MultiProbe only fork

Apprisco, 2024-10-12:
Hi! This may be impossible/against what you're going fork, but would it be possible to maintain a branch of klipper with ONLY the multi probe feature enabled?
https://klipper.discourse.group/t/klipper-for-cnc-initiatives-and-projects-list/5698/121

A big issue with klipper is that it was baked with only one probe in mind, but multiple probes let you do crazy things like *build a stealthchanger with a cartographer*.
Stealthchanger uses TAP, while Carto is faster in bed meshing and has crazy good survey touch.
These would be an excellent option to add to Klipper, and just the multiple probe functionality already creates a TON of improvements over base.

Thank you!

noobydp, 2024-10-12:
I've been wondering if they'd let us merge the multi probe feature into mainline Klipper. 

I'd like multiple probes too.

naikymen, 2024-10-12:
Hello! Thanks for the praise on the forum. :)

It is currently less work to merge upstream regularly, than to spend an indefinite amount of time merging it upstream. At the moment this fork is up to date, and is easy to install and update with KIAUH and Mainsail.

@Apprisco @noobydp what are your motivations? is there something missing in this fork that you need or that is malfunctioning?

I could try putting together a PR for the multiple probes code, but I really don't want to waste any time with the "review process". I could help but someone else should lead that part.

Kevin has shown interest in this feature, as part of hist objectives for 2024 [here](https://klipper.discourse.group/t/klipper-development-goals-for-2024/12808), along with a rework of the homing and probing code.

However, I'm not sure Kevin would like the way I implemented multiple probe support; I did what I could, and he would probably do it better. What may end up happening, as has been seen with Cirromulus' work, is that the the feature is re-implemented from scratch, in a "better" way.

Apprisco, 2024-10-12:
I don't mind it being a separate fork rather than a push into Klipper main! It'd just be very useful to have a fork of klipper that supports multiple probes. It's a pretty niche need in the first place.

For example, TAP and Carto. TAP is incredibly accurate, while carto can do incredibly fast bed meshes.
Mixing multiple probes together would lead to the best of both worlds.

My goal is different. Stealthchanger toolhead changers require TAP as part of their codebase, but carto is preferable for big meshes. As such, mixing the two would be great.

a PR would be cool, but just a way to implement multiple probes only as part of this fork would be great :)
I got the klipper-cnc fork working with BLTouch and Cartographer which was interesting :D, and just a way to ONLY implement multiple probes would be awesome.

I understand this isn't what you made this branch for, but it's really useful!

naikymen, 2024-10-13:
Hmm I think I don't understand what you need. I'm not familiar with your hardware, you may need yo ELI5 me on this one.

> It'd just be very useful to have a fork of klipper that supports multiple probes.

It seems to me that this fork already does that. Or maybe what I meant by multiple probes is no the same as what you're describing. Multiple probes, as I've been considering them, refers to having more than one probe available for G38-like probing moves.

What do you mean by "only" implementing multiple probes exactly?

Apprisco, 2024-10-13:
This fork DOES achieve it, but it also adds a lot of other "unnecessary" functionality. I'd ilke to keep the changes from base klipper minimal if possible.
That's exactly what it means, having multiple probes you can choose from.
However, there's many other configuration changes and features the CNC fork adds that wouldn't be needed for pure multi-probe toolchanging on a generic 3D printer, as such I was wondering if you could isolate the module for multiple probes like the klippy module.

noobydp, 2024-10-15:
Hey Nic,

> @noobydp what are your motivations? is there something missing in this fork that you need or that is malfunctioning?

I didnt meant to imply there's anything wrong with this fork :)

I use this on my actual CNC which has several probes and everything works really great.

I recently built a 3d printer with a tool changer which was 'inspired' by the [Proforge 4](https://www.makertech3d.com/pages/proforge-4).
While I was building it, I thought it would be nice to have a probe per toolhead, plus another probe for calibrating x y offsets.
In the end I used a single bl touch probe, and this plugin which adds support for a single multi-axis probe for calculating offsets of each toohead https://github.com/viesturz/klipper-toolchanger

naikymen, 2024-10-15:
Thanks for explaining!

I don't think I will work on this as I am focusing on other projects, and this does not involve fixing an issue nor adding features for a new use case.

Now I understand what you meant by "against what you're going for".

I will close this for now, but I would still like to know why this request has come up at all. Why is this important to you really?

The only thing that you can't disable as of now is the multi-axis feature, which is already being used by advanced 3D-printer prototypes. It's not making sense to me at all why you would call this unnecessary, or why removing it would make it more "pure" at all.

closed 2024-10-15 by naikymen

naikymen, 2024-10-15:
There is also the KTC project: https://github.com/TypQxQ/KTC

Have you tried it out? I confess that I don't understand very well what it does.

noobydp, 2024-10-16:
I tried KTC first but it wasn't as good as the toolchanger project I linked above. 

I think it didn't have any expanded probe support, and was more difficult to configure in comparison.

Apprisco, 2024-10-16:
hey @naikymen, wondering about carto vs TAP.
Is it possible to add on cartographer probe as the extra probe? Or TAP?
Both of them are different than traditional probes such as Klicky or BLTouch.

Thanks!

naikymen, 2024-10-18:
>  Is it possible to add on cartographer probe as the extra probe? Or TAP?
> Is it possible to add on cartographer probe as the extra probe? Or TAP?
> Both of them are different than traditional probes such as Klicky or BLTouch.

I'm sorry but I don't have access to this hardware, so I don't know. If you can summarize the differences and point me the relevant sources I can check it out.

I've not tried [`klipper-toolchanger`](https://github.com/viesturz/klipper-toolchanger) yet. I'm not sure how it works, how it is used, nor what it needs from a probe object to work.

I'd like to know if viesturz is interested in helping out. They could give us jump-start, otherwise I'd be spelunking through klipper code (yet again) and I'm not thrilled about it unfortunately. :/

> Tried for ~10 hours and lost my marbles.

Have you documented your efforts?

If you want me to help with any of these things, feel free to open issues here and document your progress. I'll try to be useful with what I know.