naikymen/klipper-for-cnc #32: Extruder home command "not existing" when using none kinematics

DJProPlusMax, 2024-10-23:
Hi, I am building a thing with just one motor, and to do that in Klipper, I need to use "none" kinematics. Unfortunately, when i try to home using either the "HOME_EXTRUDER EXTRUDER=extruder" or "HOME_ACTIVE_EXTRUDER" command. I made sure to use the Klipper for CNC repo and recompiled Klipper for the MCU. When I took a look at the none.py file, I couldn't see any section for the homeable extruder - but I did notice such section in the extruder.py file. Can this be solved somehow?
Thanks in advance
Dan

* Klipper log file: [klippy (1).log](https://github.com/user-attachments/files/17489266/klippy.1.log)
* Klipper config files:  [printer.txt](https://github.com/user-attachments/files/17489289/printer.txt) - had to change to TXT cuz Github
* 
[image]

naikymen, 2024-10-23:
Hi DJProPlusMax! It seems that you have not configured extruder homing.

By default this fork behaves just as mainline klipper, so you need to configure the extruder for homing, as exemplified in the main README file.

https://github.com/naikymen/klipper-for-cnc?tab=readme-ov-file#klipper-for-cnc

Have you had a look already?

DJProPlusMax, 2024-10-23:
Hi,
Sorry for my incompetence reading setup manuals. I completely forgot the [EXTRUDER_HOME extruder] section, so tommorow I’ll correct this and my machine should work as I want it to work.

I once more apologize for my incompetence reading manuals and any time loss incurred by my mistake.
Have a nice day
Dan

DJProPlusMax, 2024-10-24:
@naikymen 
Well, I added the section, it recognizes it now but it says internal error. I tried both  commands. What is happening?
[klippy.log](https://github.com/user-attachments/files/17503793/klippy.log)
[image]

DJProPlusMax, 2024-10-26:
@naikymen 
How is it going? Is it just an issue of the homeable extruder section missing from none kinematics, or is it something more?
Have a nice day
Dan

Jarewa, 2024-10-27:
I'm also curious, where can I find the example configuration file "home_extruder.cfg"??

DJProPlusMax, 2024-10-27:
@Jarewa 
Hi, there is no such thing as “home_extruder.cfg”, there is however “[extruder.py](https://github.com/naikymen/klipper-for-cnc/blob/pipetting/klippy/kinematics/extruder.py)” and that has a homeable extruder section. I think my problem lies in the fact that the homeable extruder thing was ported to cartesian kinematics, but not “none” kinematics.

naikymen, 2024-10-27:
Hi again!

@Jarewa  there are no configuration parameters for the `[home_extruder]` sections. Simply adding one for an extruder will register commands for homing it.

@DJProPlusMax I haven't had time yet to investigate. Hopfully I'll have time today or tomorrow.

naikymen, 2024-10-27:
Here is a quick and dirty patch: https://github.com/naikymen/klipper-for-cnc/commit/be4c9287c539ab92b28ad47d63eee6357a4f5e95

The fact is that I have not adapted the "None" kinematics to work well with the 6-axis stuff. I will do eventually.

Let me know how it goes!

DJProPlusMax, 2024-10-28:
@naikymen 
Hi, so I just tried installing it - updating via Moonraker and then reflashing the MCU, no dice...
[klippy after be4c928.log](https://github.com/user-attachments/files/17541035/klippy.after.be4c928.log)
[image]

naikymen, 2024-11-20:
Hi @DJProPlusMax thanks for reporting. I just pushed a few fixes for motion_report.

Please let me know if they work, I won't have time to test the setup this week.