naikymen/klipper-for-cnc #33: XY movement only on trigger

PaddyCube, 2024-12-12:
**Is your feature request related to a problem? Please describe.**
I'm building a embroidery machine. XY movement must only occur, if needle of sewing machine is in upper most position. There is an optical sensor which triggers if in upper position.

GCode itself consists of XY moves only. All Z-moves (needle) are separate moves like:

G0 X1 Y1
G0 Z1
G0 X1.1 Y1
G0 Z2
G0 X1.1 Y1.1
G0 Z3
...

**Describe the solution you'd like**
I need an option which ensures XY movement only occurs when needle sensor triggers. Each G0 and G1 gets confirmed by the trigger and only after trigger (needle upmost position), next command gets executed.

As I don't want to stop sewing machine after each stich and there is only a short time between needle up and needle down, only short, small XY movements can be done between. If machine needs a movement larger than a given threshold, needle first must be stopped completely in upper pos before travelling.

**Describe alternatives you've considered**
As an alternative I might can provide G Code with G0 for travel and G1 for stiching. G1 is only allowed when needle triggers but no stop of needle is necessary. G0 is only allowed if needle triggers AND machine motor halts.

jakobwowy, 2024-12-14:
You could use gcode button to handle the trigger on the upper position. Then you can write a saved variable on the gcode buttion actions.  For the movements you could create a macro which check if the saved variable has the right value and only execute the movements when the value is set.

naikymen, 2024-12-15:
Hi I tend to line up with @jakobwowy's approach. Perhaps an override of the G1 command, plus a similar check for the safety pin, may work. I'm no macro wizard though, I don't know if this would be fast enough.

An alternative would be using probes, but I don't think that the new probe code would help in this case. This idea of "waiting" for something and moving immediately after (and fast enough, before the needle comes down again) does not look compatible with how probing works in klipper.

Overall the idea of synchronizing the motion of a PID-controlled DC motor with that of the stepper motors seems hard. 

If you are willing to invest in hardware, another way to do it would be to add an encoder to the DC motor, and put it behind a closed-loop diver that accepts STEP-DIR pulses as an input (which klipper understands). You might find the SimpleFOC project useful in this case: https://docs.simplefoc.com/step_dir_interface

naikymen, 2024-12-15:
Perhaps a custom firmware implementation would be better in this case.

What are your reasons to use klipper for your project in particular?

PaddyCube, 2024-12-15:
I'm using klipper for my printer and thought this might be a good idea to use it for this project as well. There is only one firmware out there which is almost 6 yrs old and not maintained anymore. Klipper has a great community and support which leads to my choice.

I'm still digging into Code to understand how klipper works. What I don't understand so far is how gets a gcode file streamed? Does klipper reads the file and process line after line? Or does Mainsail read the file and send each line individually to klipper api?

naikymen, 2024-12-16:
You may find answers for these questions in the official docs https://www.klipper3d.org/Overview.html and the forum's knowledgebase https://klipper.discourse.group/c/knowledge-base/16

And now the usual warning for new folks: Klipper is a 3D printer firmware. Period.

I don't mean to be harsh, but there are several design decisions (and owner roadblocks) that make it _hard_ to adapt to new applications. I have experienced this.

If you are not determined to use Klipper, and there is a specific open-source project you can revive, I'd go for it. Others may thank you for it.

There are also other modern and more general firmware projects for CNC that may be much better for you, for example:  [FluidNC](https://github.com/bdring/FluidNC), [uCNC](https://github.com/Paciente8159/uCNC), [RRF](https://docs.duet3d.com/User_manual/RepRapFirmware/RepRapFirmware_overview), [grblHAL](https://github.com/grblHAL), and so on.

PaddyCube, 2024-12-17:
Thanks to all of you to your valuable input. I really appreciate it.
I'll try to build a step/dir interface for dc motor. Shouldn't be that hard. Also thanks for the gcode_button hint. 

If it doesn't work at all I try to evolve the old firmware I found named teathimble

closed 2024-12-17 by PaddyCube